wpilibsuite/GradleRIO #657: DefaultHeaderDependenciesCollector issue with wpilib/opencv headers

dstalter4, 2023-04-16:
Dependency detection does not work correctly resulting in excessively long build times, as the build process assumes all translation units need to be rebuilt.  When robot programs have a higher numbers of translation units, this can result in minutes worth of building when most object files may be up to date already.  Old/slow build PC hardware can exacerbate the problem.

The issue stems from headers inside wpilib/opencv using sequences like the following:

#ifdef EIGEN_FUNCTORS_PLUGIN
#include EIGEN_FUNCTORS_PLUGIN
#endif

In the case where the macros aren't defined, the expansion is empty, leading to gradle not knowing how to handle the #include line.  Here are a few of examples:

https://github.com/wpilibsuite/allwpilib/blob/9b1bf5c7f1f624dd7ebd0f9cce1f0f178a95ce8c/wpimath/src/main/native/thirdparty/eigen/include/Eigen/src/Core/functors/StlFunctors.h#L158
https://github.com/wpilibsuite/allwpilib/blob/9b1bf5c7f1f624dd7ebd0f9cce1f0f178a95ce8c/wpimath/src/main/native/thirdparty/eigen/include/Eigen/src/Core/DenseBase.h#L660
https://github.com/wpilibsuite/allwpilib/blob/9b1bf5c7f1f624dd7ebd0f9cce1f0f178a95ce8c/wpimath/src/main/native/thirdparty/eigen/include/Eigen/src/Core/Matrix.h#L441

The opencv headers have the same problem with macros OPENCV_INCLUDE_PORT_FILE and OPENCV_STDINT_HEADER.

Sample output from .\gradlew build --debug

2023-04-15T21:51:40.862-0400 [INFO] [org.gradle.language.nativeplatform.internal.incremental.IncrementalCompileFilesFactory] Cannot locate header file for '#include EIGEN_FUNCTORS_PLUGIN' in source file 'StlFunctors.h'. Assuming changed.
2023-04-15T21:51:40.862-0400 [INFO] [org.gradle.language.nativeplatform.internal.incremental.IncrementalCompileFilesFactory] Cannot locate header file for '#include EIGEN_DENSEBASE_PLUGIN' in source file 'DenseBase.h'. Assuming changed.
2023-04-15T21:51:40.862-0400 [INFO] [org.gradle.language.nativeplatform.internal.incremental.IncrementalCompileFilesFactory] Cannot locate header file for '#include EIGEN_MATRIXBASE_PLUGIN' in source file 'MatrixBase.h'. Assuming changed.
2023-04-15T21:51:40.862-0400 [INFO] [org.gradle.language.nativeplatform.internal.incremental.IncrementalCompileFilesFactory] Cannot locate header file for '#include EIGEN_ARRAYBASE_PLUGIN' in source file 'ArrayBase.h'. Assuming changed.
2023-04-15T21:51:40.863-0400 [INFO] [org.gradle.language.nativeplatform.internal.incremental.IncrementalCompileFilesFactory] Cannot locate header file for '#include EIGEN_PLAINOBJECTBASE_PLUGIN' in source file 'PlainObjectBase.h'. Assuming changed.
2023-04-15T21:51:40.863-0400 [INFO] [org.gradle.language.nativeplatform.internal.incremental.IncrementalCompileFilesFactory] Cannot locate header file for '#include EIGEN_MATRIX_PLUGIN' in source file 'Matrix.h'. Assuming changed.
2023-04-15T21:51:40.863-0400 [INFO] [org.gradle.language.nativeplatform.internal.incremental.IncrementalCompileFilesFactory] Cannot locate header file for '#include EIGEN_ARRAY_PLUGIN' in source file 'Array.h'. Assuming changed.
2023-04-15T21:51:40.863-0400 [INFO] [org.gradle.language.nativeplatform.internal.incremental.IncrementalCompileFilesFactory] Cannot locate header file for '#include EIGEN_MAPBASE_PLUGIN' in source file 'MapBase.h'. Assuming changed.
2023-04-15T21:51:40.863-0400 [INFO] [org.gradle.language.nativeplatform.internal.incremental.IncrementalCompileFilesFactory] Cannot locate header file for '#include EIGEN_VECTORWISEOP_PLUGIN' in source file 'VectorwiseOp.h'. Assuming changed.
2023-04-15T21:51:40.865-0400 [DEBUG] [org.gradle.internal.operations.DefaultBuildOperationRunner] Build operation 'Processing source files' completed
2023-04-15T21:51:40.865-0400 [INFO] [org.gradle.language.nativeplatform.internal.incremental.DefaultHeaderDependenciesCollector] After parsing the source files, Gradle cannot calculate the exact set of include files for ':compileFrcUserProgramDebugExecutableFrcUserProgramCpp'. Every file in the include search path will be considered a header dependency.
2023-04-15T21:51:40.865-0400 [INFO] [org.gradle.language.nativeplatform.internal.incremental.DefaultHeaderDependenciesCollector] adding files in C:\Repos\FRC120\YtaRobot\src\main\include to header dependencies for :compileFrcUserProgramDebugExecutableFrcUserProgramCpp
2023-04-15T21:51:40.865-0400 [INFO] [org.gradle.language.nativeplatform.internal.incremental.DefaultHeaderDependenciesCollector] adding files in C:\Users\FRC\.gradle\caches\transforms-3\9c6bb2103a3d82450f87644c2c782003\transformed\wpilibNewCommands-cpp-2023.4.3-headers to header dependencies for :compileFrcUserProgramDebugExecutableFrcUserProgramCpp
2023-04-15T21:51:40.868-0400 [INFO] [org.gradle.language.nativeplatform.internal.incremental.DefaultHeaderDependenciesCollector] adding files in C:\Users\FRC\.gradle\caches\transforms-3\ae965e1859d74c01db2b3fd8527e3c6f\transformed\REVLib-cpp-2023.1.3-headers to header dependencies for :compileFrcUserProgramDebugExecutableFrcUserProgramCpp
2023-04-15T21:51:40.868-0400 [INFO] [org.gradle.language.nativeplatform.internal.incremental.DefaultHeaderDependenciesCollector] adding files in C:\Users\FRC\.gradle\caches\transforms-3\16c2e4ca63c776cbdd6f665d62feede4\transformed\REVLib-driver-2023.1.3-headers to header dependencies for :compileFrcUserProgramDebugExecutableFrcUserProgramCpp
2023-04-15T21:51:40.868-0400 [INFO] [org.gradle.language.nativeplatform.internal.incremental.DefaultHeaderDependenciesCollector] adding files in C:\Users\FRC\.gradle\caches\transforms-3\9bb6607890a98171a5d27614d7869828\transformed\wpiapi-cpp-5.30.4-headers to header dependencies for :compileFrcUserProgramDebugExecutableFrcUserProgramCpp
2023-04-15T21:51:40.870-0400 [INFO] [org.gradle.language.nativeplatform.internal.incremental.DefaultHeaderDependenciesCollector] adding files in C:\Users\FRC\.gradle\caches\transforms-3\a275bec10651059ff2d203c4c1a9eab9\transformed\api-cpp-5.30.4-headers to header dependencies for :compileFrcUserProgramDebugExecutableFrcUserProgramCpp
2023-04-15T21:51:40.872-0400 [INFO] [org.gradle.language.nativeplatform.internal.incremental.DefaultHeaderDependenciesCollector] adding files in C:\Users\FRC\.gradle\caches\transforms-3\c79a9f0154dbfa65398a6698d022290d\transformed\cci-5.30.4-headers to header dependencies for :compileFrcUserProgramDebugExecutableFrcUserProgramCpp
2023-04-15T21:51:40.875-0400 [INFO] [org.gradle.language.nativeplatform.internal.incremental.DefaultHeaderDependenciesCollector] adding files in C:\Users\FRC\.gradle\caches\transforms-3\e06242a3798af35219d5658561f632ac\transformed\tools-23.0.10-headers to header dependencies for :compileFrcUserProgramDebugExecutableFrcUserProgramCpp
2023-04-15T21:51:40.879-0400 [INFO] [org.gradle.language.nativeplatform.internal.incremental.DefaultHeaderDependenciesCollector] adding files in C:\Users\FRC\.gradle\caches\transforms-3\d36a5063d14d66f4cf78f633d073966e\transformed\wpilibc-cpp-2023.4.3-headers to header dependencies for :compileFrcUserProgramDebugExecutableFrcUserProgramCpp
2023-04-15T21:51:40.883-0400 [INFO] [org.gradle.language.nativeplatform.internal.incremental.DefaultHeaderDependenciesCollector] adding files in C:\Users\FRC\.gradle\caches\transforms-3\6de908f44183828b88b1804786e77788\transformed\ntcore-cpp-2023.4.3-headers to header dependencies for :compileFrcUserProgramDebugExecutableFrcUserProgramCpp
2023-04-15T21:51:40.884-0400 [INFO] [org.gradle.language.nativeplatform.internal.incremental.DefaultHeaderDependenciesCollector] adding files in C:\Users\FRC\.gradle\caches\transforms-3\cbbaac75b20bb565dca5e8089c410db2\transformed\hal-cpp-2023.4.3-headers to header dependencies for :compileFrcUserProgramDebugExecutableFrcUserProgramCpp
2023-04-15T21:51:40.885-0400 [INFO] [org.gradle.language.nativeplatform.internal.incremental.DefaultHeaderDependenciesCollector] adding files in C:\Users\FRC\.gradle\caches\transforms-3\062826165847ba6b02017b06274bdccb\transformed\wpimath-cpp-2023.4.3-headers to header dependencies for :compileFrcUserProgramDebugExecutableFrcUserProgramCpp
2023-04-15T21:51:40.898-0400 [INFO] [org.gradle.language.nativeplatform.internal.incremental.DefaultHeaderDependenciesCollector] adding files in C:\Users\FRC\.gradle\caches\transforms-3\f4c52b32b5e3a607487b45280649e26c\transformed\wpinet-cpp-2023.4.3-headers to header dependencies for :compileFrcUserProgramDebugExecutableFrcUserProgramCpp
[... cut ...]
2023-04-15T21:51:40.930-0400 [INFO] [org.gradle.language.nativeplatform.internal.incremental.DefaultHeaderDependenciesCollector] adding files in C:\Users\Public\wpilib\2023\roborio\arm-nilrt-linux-gnueabi\sysroot\usr\include to header dependencies for :compileFrcUserProgramDebugExecutableFrcUserProgramCpp
2023-04-15T21:51:41.047-0400 [DEBUG] [org.gradle.internal.operations.DefaultBuildOperationRunner] Build operation 'Snapshot inputs and outputs before executing task ':compileFrcUserProgramDebugExecutableFrcUserProgramCpp'' completed

ThadHouse, 2023-04-16:
I had thought we had the workaround for this for robot projects, but it looks like its only actually in place for the allwpilib build. 

You can work around this now by adding the following code to `settings.gradle`

```
Properties props = System.getProperties();
props.setProperty("org.gradle.internal.native.headers.unresolved.dependencies.ignore", "true");
```

dstalter4, 2023-04-16:
Thanks, @ThadHouse.  I tried your suggestion and can confirm it does fix the issue when placed into the robot project's settings.gradle.

closed 2023-09-01 by PeterJohnson