wpilibsuite/GradleRIO #674: GradleRIO doesn't install vendordeps for C++ projects on Linux

spacey-sooty, 2023-10-08:
We our setting up our repository for 2024 and some of our team members this year use Linux. We are a long time C++ team with a lot of libraries written in C++ so we don't see switching to Java (which we have not encountered this issue with) as a viable option. These are the error logs for our CI pipeline using the ubuntu docker image for WPIlib.
```
> Task :wombat:compileWombatLinuxathenaDebugStaticLibraryWombatCpp
In file included from /__w/2024-Crescendo/2024-Crescendo/wombat/src/main/cpp/utils/Encoder.cpp:1:0:
/__w/2024-Crescendo/2024-Crescendo/wombat/src/main/include/utils/Encoder.h:10:10: fatal error: ctre/phoenix.h: No such file or directory
 #include <ctre/phoenix.h>
```
https://github.com/spacey-sooty/2024-Crescendo/tree/ci
When we look through the error logs we see that all calls to `ctre/phoenix` and `frc/geometry/Pose3d.h` give an error but other includes do not. We have confirmed that our code builds on Windows and we get the same error message when we try building on Linux Desktops. 

We suspect that GradleRIO isn't installing vendordeps or installing them incorrectly. Any help would be greatly appreciated.

ThadHouse, 2023-10-08:
That CI branch is building against 2022 wpilib (`id "edu.wpi.first.GradleRIO" version "2022.2.1")`, which doesn't have Pose3d. So that explains that error. The phoenix one I think is due to linux being case sensitive, and the casing for that phoenix include is probably wrong. Windows doesn't care, which would cause it to work fine there.

spacey-sooty, 2023-10-08:
Hi @ThadHouse thanks for the feedback! It looked like we did have some of those problems I have switched onto a mentors computer and fixed up all of the issues with old versions etc. I have now double checked that it definitely builds on windows with updated versions but we still get the issue in CI over `ctre/phoenix`. Everything worked fine even without most things installed as it was a new computer and GradleRIO installed everything properly on windows but on Linux and in CI we are still getting failing builds. Could there be anything to do with say public directory configuration that we would need to add on to fix this on Linux? If not I believe this would be an issue with GradleRIO's downloads. Thanks again for your help!

spacey-sooty, 2023-10-08:
@ThadHouse it was because of Linux being case sensitive thanks for your help!

closed 2023-10-08 by spacey-sooty