wpilibsuite/GradleRIO #677: Error out early if building on OneDrive

PeterJohnson, 2023-10-13:
OneDrive breaks caching, so we should check early if the project is on OneDrive and error out if it is.

spacey-sooty, 2023-10-18:
Hey @PeterJohnson  I can try and implement this, I am currently thinking to take the full directory of the project and check if OneDrive is a string within it. Does this seem good or is there a different implementation that you had in mind?

PeterJohnson, 2023-10-18:
That's pretty much exactly what I was thinking.  It's unlikely that will hit false positives (e.g. someone names their user or project "OneDrive" feels unlikely).

spacey-sooty, 2023-10-19:
Hey @PeterJohnson  I thought something like that was unlikely as well but we might want to add a note somewhere in the documentation about not creating projects in OneDrive. Not 100% sure where we would put that but I'm more than happy to add a note to the docs if you tell me where to

PeterJohnson, 2023-10-19:
That’s why there’s also a companion issue to this one on vscode-wpilib. If we do the same detection there it can warn when opening or prevent creation of a project, which should cover most users.

spacey-sooty, 2023-10-19:
@PeterJohnson that sounds good I hadnt seen that, I would guess most people not creating their project using the vscode creator would be keeping track of issues here

PR #680 (2023-11-03): Fix #677 Error if project is in OneDrive
Should Resolve #677
commit e05c24fb (2023-11-03): Fix #677 Error if project is in OneDrive (#680)

* search project initialised directory for onedrive in execute function

* apply changes from review

* fix error being supressed

* use runtime exception in onedrive error

* add print in error and make check case insensitive

* use Paths.get instead of user.dir

* move check to apply method and use gradle method instead of java method

* remove unnecessary imports from previous commit

* remove another unnecessary import

* fix printed directory

* catch and print error instead of throwing it

* try catch check only throw if try catch succeeds [... cut]
closed 2023-11-03 by ThadHouse