wpilibsuite/GradleRIO #687: Inconsistent simulation builds

spacey-sooty, 2023-11-24:
**Describe the bug**
When running `./gradlew simulateNativeDebug`  I get inconsistent results with the same code. 

**To Reproduce**
Steps to reproduce the behavior:
1. run `git clone https://github.com/spacey-sooty/2024-Crescendo.git`
2. run `cd 2024-Crescendo`
3. run `./gradlew simulateNativeDebug`
4. See this error message
```
> Task :wombat:simulateNativeDebug FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':wombat:simulateNativeDebug'.
> Must have 1 and only 1 binary

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

BUILD FAILED in 16s
3 actionable tasks: 3 executed
```
6.  run `./gradlew build` (succeeds)
7. run `./gradlew simulateNative`
8. See this error message
```
> Task :wombat:simulateNativeRelease FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':wombat:simulateNativeRelease'.
> Must have 1 and only 1 binary

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

BUILD FAILED in 12s
6 actionable tasks: 2 executed, 4 up-to-date
```
9. repeat steps 7 and 8 once.
10. run `./gradlew simulateNativeDebug`
11. Success

 - Link to code: https://github.com/spacey-sooty/2024-Crescendo/tree/master

**Expected behavior**
I expect when I simulate my project with no changes to it the result is the same.

**Desktop (please complete the following information):**
 - OS: Linux Mint 21.2
 - Project Information: 
[image]

ThadHouse, 2023-11-24:
The main issue with this is we don't support gradlerio being loaded in multiple projects at the same time. What is happening is the sim tasks are being created for the Wombat subproject, but because its not an executable it doesn't get configured right. 

I'll have to look to see if I can find a way to solve this, but due to how close it is to the season, and it being an unsupported scenario I'm not sure we'll get to it for 2024.

ThadHouse, 2023-11-24:
As a workaround, you can explicitly run the root tasks `:simulateNativeRelease` and `:simulateNativeDebug` and you won't see this.

spacey-sooty, 2023-11-24:
Ok thank you for the temporary work around.

spacey-sooty, 2023-11-24:
> The main issue with this is we don't support gradlerio being loaded in multiple projects at the same time. What is happening is the sim tasks are being created for the Wombat subproject, but because its not an executable it doesn't get configured right.
> 
> I'll have to look to see if I can find a way to solve this, but due to how close it is to the season, and it being an unsupported scenario I'm not sure we'll get to it for 2024. And any fix for this would likely be a project breaking change.

I'll also take a look see if I can find a way to resolve this, preferably without huge changes