wpilibsuite/GradleRIO #697: ./gradlew generateCompileCommands outputs no compile_command.json

spacey-sooty, 2023-12-11:


ThadHouse, 2023-12-11:
It generates into the build folder, but the location will be changing again in the next beta release and for the 2024 season.

closed 2023-12-11 by ThadHouse

spacey-sooty, 2023-12-11:
> It generates into the build folder, but the location will be changing again in the next beta release and for the 2024 season.
> 
> It will generate into `build/TargetedCompileCommands/targetplatformandbuildtype`.

This isn't documented, could it be worth to mention it in FRC docs?

ThadHouse, 2023-12-11:
The only part is we don't want to recommend compile_commands for general users. There are a lot of pitfalls that many teams wouldn't know about, that the built in intellisense doesn't have.

spacey-sooty, 2023-12-11:
Might be worth to do it but with a warning. It is under advanced GradleRIO either way.