wpilibsuite/GradleRIO #711: Glass and SysId not installed via installAllTools

person4268, 2024-01-07:
This is because:
a) https://github.com/wpilibsuite/GradleRIO/blob/c7a425a39dc479f3528a3737ec65d7d5445c2d4f/src/main/java/edu/wpi/first/gradlerio/wpi/dependencies/tools/WPIToolsPlugin.java#L54-L61 doesn't iterate over the cppTools array

b) https://github.com/wpilibsuite/GradleRIO/blob/c7a425a39dc479f3528a3737ec65d7d5445c2d4f/src/main/java/edu/wpi/first/gradlerio/wpi/dependencies/tools/WPICppTool.java#L30 is commented out

and c) WPICppTool has no getToolInstallTask() method (not that it would be called without a being fixed.

Not sure whether it's intentional as the git history doesn't say much, but I believe I asked Peter (iirc) at Worlds and he said it wasn't intentional and was a bug (the specific question I asked was why `installAllTools` on Linux didn't install Glass or SysID, but it seems like it doesn't install it at all).

ThadHouse, 2024-01-08:
Its not _intentional_, its more that I just haven't had time. Its a little more complicated to grab the C++ tools on windows, as we also need to grab the MSVC runtime.

person4268, 2024-01-08:
Would bailing out/warning on Windows if the prerequisite runtime (not sure specifically which year MSVC runtime you need) is not installed be sufficient? Or are you looking more towards getting them installed (perhaps by invoking the installer or getting the MSVC dlls somehow and putting them next to the exes).

ThadHouse, 2024-01-08:
We install the runtime DLLs next to the exe when the installer runs. So that side works. So it's just gradle that we need to implement that part. The gradle tool download is not our recommendation however, we very much do recommend the installer.

person4268, 2024-01-08:
I do agree that it's generally the best option, but I generally prefer to use the gradle tool download on my personal machine, as I don't have much use for the offline artifacts and won't keep it up to date relative to my robot project anyway. I'll look towards me implementing it the coming couple of weeks if I find time to. Seems like I'd need to make gradle download edu.wpi.first.msvc:runtime from maven (if on windows) and create a CppToolInstallTask that includes all the necessary logic.