wpilibsuite/GradleRIO #727: Error `> Failed to apply plugin class 'edu.wpi.first.gradlerio.wpi.WPIPlugin'.`

spacey-sooty, 2024-02-28:
Full error message:
```
FAILURE: Build failed with an exception.

* Where:
Build file '/home/spacey_sooty/Documents/frc/2024-Crescendo/build.gradle' line: 4

* What went wrong:
An exception occurred applying plugin request [id: 'edu.wpi.first.GradleRIO', version: '2024.3.1']
> Failed to apply plugin class 'edu.wpi.first.gradlerio.wpi.WPIPlugin'.
   > Could not create an instance of type edu.wpi.first.gradlerio.wpi.WPIExtension.
      > Cannot read the array length because "<local4>" is null
```
I am getting this error when trying to build my project. I believe it is due to one of my vendordeps being invalid but this error message was rather unhelpful. I'm not sure how this could be improved as I'm not 100% sure what the error is... I can send through info about the vendordeps file etc if required.

PeterJohnson, 2024-02-28:
Please run gradlew with the —-stacktrace option.  And yes, posting a copy of your project including vendordeps would enable us to try to reproduce.  Does the build failure only happen on one computer?

spacey-sooty, 2024-02-28:
Haven't tested on other computers. Here is the failing project https://github.com/spacey-sooty/VendordepTesting
here is the project the vendordep is from (I know maven isn't properly published I was trying to test the json file before getting that all working) https://github.com/spacey-sooty/Numbat 
```

FAILURE: Build failed with an exception.

* Where:
Build file '/home/spacey_sooty/Documents/frc/NumbatInstallTest/build.gradle' line: 4

* What went wrong:
An exception occurred applying plugin request [id: 'edu.wpi.first.GradleRIO', version: '2024.3.1']
> Failed to apply plugin class 'edu.wpi.first.gradlerio.wpi.WPIPlugin'.
   > Could not create an instance of type edu.wpi.first.gradlerio.wpi.WPIExtension.
      > Cannot read the array length because "<local4>" is null

* Try:
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

* Exception is:
org.gradle.api.plugins.InvalidPluginException: An exception occurred applying plugin request [id: 'edu.wpi.first.GradleRIO', version: '2024.3.1']
        at org.gradle.plugin.use.internal.DefaultPluginRequestApplicator.exceptionOccurred(DefaultPluginRequestApplicator.java:260)
        at org.gradle.plugin.use.internal.DefaultPluginRequestApplicator.applyPlugin(DefaultPluginRequestApplicator.java:238)
        at org.gradle.plugin.use.internal.DefaultPluginRequestApplicator.applyLegacyPlugin(DefaultPluginRequestApplicator.java:190)
        at org.gradle.plugin.use.internal.DefaultPluginRequestApplicator.access$300(DefaultPluginRequestApplicator.java:63)
        at org.gradle.plugin.use.internal.DefaultPluginRequestApplicator$1$1.lambda$addLegacy$0(DefaultPluginRequestApplicator.java:150)
        at org.gradle.plugin.use.internal.DefaultPluginRequestApplicator.lambda$applyPlugins$0(DefaultPluginRequestApplicator.java:108)
        at org.gradle.plugin.use.internal.DefaultPluginRequestApplicator.applyPlugins(DefaultPluginRequestApplicator.java:108)
        at org.gradle.configuration.DefaultScriptPluginFactory$ScriptPluginImpl.apply(DefaultScriptPluginFactory.java:119)
        at org.gradle.configuration.BuildOperationScriptPlugin$1.run(BuildOperationScriptPlugin.java:68)
        at org.gradle.internal.operations.DefaultBuildOperationRunner$1.execute(DefaultBuildOperationRunner.java:29)
        at org.gradle.internal.operations.DefaultBuildOperationRunner$1.execute(DefaultBuildOperationRunner.java:26)
        at org.gradle.internal.operations.DefaultBuildOperationRunner$2.execute(DefaultBuildOperationRunner.java:66)
        at org.gradle.internal.operations.DefaultBuildOperationRunner$2.execute(DefaultBuildOperationRunner.java:59)
        at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:157)
        at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:59)
        at org.gradle.internal.operations.DefaultBuildOperationRunner.run(DefaultBuildOperationRunner.java:47)
        at org.gradle.internal.operations.DefaultBuildOperationExecutor.run(DefaultBuildOperationExecutor.java:68)
        at org.gradle.configuration.BuildOperationScriptPlugin.lambda$apply$0(BuildOperationScriptPlugin.java:65)
        at org.gradle.internal.code.DefaultUserCodeApplicationContext.apply(DefaultUserCodeApplicationContext.java:43)
        at org.gradle.configuration.BuildOperationScriptPlugin.apply(BuildOperationScriptPlugin.java:65)
        at org.gradle.api.internal.project.DefaultProjectStateRegistry$ProjectStateImpl.lambda$applyToMutableState$1(DefaultProjectStateRegistry.java:406)
        at org.gradle.api.internal.project.DefaultProjectStateRegistry$ProjectStateImpl.fromMutableState(DefaultProjectStateRegistry.java:424)
        at org.gradle.api.internal.project.DefaultProjectStateRegistry$ProjectStateImpl.applyToMutableState(DefaultProjectStateRegistry.java:405)
        at org.gradle.configuration.project.BuildScriptProcessor.execute(BuildScriptProcessor.java:46)
        at org.gradle.configuration.project.BuildScriptProcessor.execute(BuildScriptProcessor.java:27)
        at org.gradle.configuration.project.ConfigureActionsProjectEvaluator.evaluate(ConfigureActionsProjectEvaluator.java:35)
        at org.gradle.configuration.project.LifecycleProjectEvaluator$EvaluateProject.lambda$run$0(LifecycleProjectEvaluator.java:109)
        at org.gradle.api.internal.project.DefaultProjectStateRegistry$ProjectStateImpl.lambda$applyToMutableState$1(DefaultProjectStateRegistry.java:406)
        at org.gradle.api.internal.project.DefaultProjectStateRegistry$ProjectStateImpl.lambda$fromMutableState$2(DefaultProjectStateRegistry.java:429)
        at org.gradle.internal.work.DefaultWorkerLeaseService.withReplacedLocks(DefaultWorkerLeaseService.java:360)
        at org.gradle.api.internal.project.DefaultProjectStateRegistry$ProjectStateImpl.fromMutableState(DefaultProjectStateRegistry.java:429)
        at org.gradle.api.internal.project.DefaultProjectStateRegistry$ProjectStateImpl.applyToMutableState(DefaultProjectStateRegistry.java:405)
        at org.gradle.configuration.project.LifecycleProjectEvaluator$EvaluateProject.run(LifecycleProjectEvaluator.java:100)
        at org.gradle.internal.operations.DefaultBuildOperationRunner$1.execute(DefaultBuildOperationRunner.java:29)
        at org.gradle.internal.operations.DefaultBuildOperationRunner$1.execute(DefaultBuildOperationRunner.java:26)
[... cut ...]
        at org.gradle.internal.instantiation.generator.AbstractClassGenerator$GeneratedClassImpl$GeneratedConstructorImpl.newInstance(AbstractClassGenerator.java:524)
        at org.gradle.internal.instantiation.generator.DependencyInjectingInstantiator.doCreate(DependencyInjectingInstantiator.java:64)
        ... 210 more

BUILD FAILED in 382ms
``` here is the stacktrace output as well. Don't know if there is an output file I could use in order to send the full stacktrace as it appears to cut off?

spacey-sooty, 2024-02-28:
I've added a CI pipeline to the repository for more reproducible builds, gives the same result I get locally as far as I can see

PeterJohnson, 2024-02-28:
This is the line that’s failing:
https://github.com/wpilibsuite/native-utils/blob/c4ed8126d6bdeabdb2575f828df0c3c7694134d9/src/main/java/edu/wpi/first/nativeutils/vendordeps/WPIVendorDepsExtension.java#L200

And it’s because your vendordep is missing a jniDependencies key.

spacey-sooty, 2024-02-28:
Hmmm ok that makes sense, if that could be documented somewhere and the error message improved that would be good. Could we add a null check here then throw a build exception if its null? I'm not sure where this could be documented. Is there some form of specification for the Vendordeps files?

spacey-sooty, 2024-02-28:
I just tested this and confirmed it does work with both of these as empty lists, thanks for the assistance! I'll take a look at providing better error messages for this tomorrow when I get some time.

PeterJohnson, 2024-02-28:
Great!  Yeah I think this could really benefit from both checking for the null case and catching/rethrowing exceptions that give more context (eg vendordep file name being processed).

closed 2024-09-17 by PeterJohnson